gradslam/gradslam #9: Building custom maps

ramilmsh, 2020-11-10:
Thank you for sharing the code! I have managed to install it successfully and run the tutorials from the tutorials branch (although some parts looked odd, but that might just be me). However, there is no reference to live feed processing yet and building custom maps and so on, please correct me if I am wrong. It would make sense for that to be in a separate package, I am just wondering if that's in the works, or should I try to add it in myself for testing purposes?

saryazdi, 2020-11-10:
Hi,

By building custom maps, do you mean using your own rgbd data? If so, that would be supported right now (although you'll need to set up your own dataloader... unless your data is in TUM format, in which case you can use the [TUM loader](https://gradslam.readthedocs.io/en/latest/modules/datasets.html#gradslam.datasets.TUM)).

As for live feed processing, we did have plans of supporting ros bags and subsccribing to ros topics, but we haven't implemented it yet. We are very interested if you want to implement that into gradslam. We are writing up the contribution guidelines right now.

ramilmsh, 2020-11-12:
@saryazdi I mean the ROS integration, I just wasn't sure whether it would be full-on ROS, ROS2 or just an interface kinda like: provide me with a stamped rgbd image, I will give you a position, so that it can later be integrated with any other data provider.

I am, unfortunately, not a free agent, but I will see if I can make time to make a PR that implements this

krrish94, 2022-12-20:
Closing due to inactivity

closed 2022-12-20 by krrish94