gradslam/gradslam #12: point cloud doesn't look that good

nawara72, 2020-11-16:
Hi
First, thanks so much for sharing your code and making it public, really appreciate it

I just installed gradslam and trying it on Tum rgbd_dataset_freiburg1_xyz. The final point cloud doesn't look that good. Would really appreciate if you can tell me what I am doing wrong. This is a snippet of the code
odometry = "gradicp"  # "gt", "icp", "gradicp"
dataset = TUM(basedir="C:/Users/cv/Documents/gradslam_data/TUM/", sequences="C:/Users/cv/Documents/gradslam_data/TUM/seq", seqlen = 10)
loader = DataLoader(dataset=dataset, batch_size=2)
colors, depths, intrinsics, poses, *_ = next(iter(loader))
rgbdimages = RGBDImages(colors, depths, intrinsics, poses, channels_first=False)

pointclouds, recovered_poses = slam(rgbdimages)
device = torch.device("cuda") if torch.cuda.is_available() else torch.device("cpu")
slam = PointFusion(odom=odometry, dsratio=4, device=device)
pointclouds, recovered_poses = slam(rgbdimages)
o3d.visualization.draw_geometries([pointclouds.open3d(0)])

Thanks

krrish94, 2020-11-16:
Hi @nawara72,

Thanks for your interest! Would it be possible for you to share a sample output reconstruction so we can assist you better?

saryazdi, 2020-11-16:
Hi @nawara72,
The `sequences` argument for `TUM` expects something like:
```
dataset = TUM(basedir="C:/Users/cv/Documents/gradslam_data/TUM/", sequences=("rgbd_dataset_freiburg1_xyz", ), seqlen = 10)
```
where you specify a tuple of sequnce names which you want to load. I'm also curious if you can share a sample output reconstruction.
Here is a working example of `PointFusion` on `TUM/rgbd_dataset_freiburg1_xyz` in Google Colab:
https://colab.research.google.com/drive/1uxjEJI5B2yJJDG3SxoAagH9uEuT67rXC?usp=sharing

nawara72, 2020-11-16:
Thanks for the quick reply and for sharing the TUM example code, really appreciate it. After running the code you shared, I got similar output to when I ran it.  I just thought the output point cloud would be more sharper (ie not so fuzzy). 
Is there any way I can write the pointcloud to a file (eg ply)  to look at it in more detail.

Thanks and thanks again for all your efforts and for sharing

saryazdi, 2020-11-16:
@nawara72 
Sure thing! I can think of 2 likely reasons for the "fuzzy"ness:
1- Might want to play around with the `point_size` of the visualization (with `open3d`, you can change point size interactively with +/- shortcuts in the visualization; with`plotly`, you can specify point size by `pointclouds.plotly(0, point_size=2)`)
2- In case of `plotly` visualization, we downsample the pointclouds before visualization because plotly visualization can be slow/break with large pointclouds.

You can use `open3d` for saving a pointcloud:
```
import open3d as o3d

o3d.io.write_point_cloud("pointcloud.ply", pointclouds.open3d(0))
```

nawara72, 2020-11-16:
Great will try that soon 👍 
Also hoping soon to try and save some rgbd data from realsense and pass it to gradslam. Do you envisage any issues with that?
Thanks again

saryazdi, 2020-11-16:
Excited to see what that will look like! Can't envisage any issues at the moment, but if you find any bugs/unexpected behavior with the library or can think of extra features which would help your use case, feel free to open up another issue and we'll look into it!

nawara72, 2020-11-16:
Great thanks for all your great support and help and will definitely share  any output that I get from realsense
Thanks and take care

closed 2020-11-16 by nawara72