gradslam/gradslam #18: Running gradslam at higher processing speed

apoorvkh, 2020-11-20:
## ❓ Questions and Help

I am only able to get ~8.5 FPS when running gradslam step-by-step on GPU. I am wondering if it's possible to run gradslam at a higher processing speed (closer to 30 FPS). I need to use a batch size of 1, sequence length of 1, and width/height of 300.

Please see the example code below (To Reproduce). I have tried lowering the ``numiters`` parameter in SLAM, and that increases FPS to 40, but I notice significant degradation in quality when I visualize the pointcloud.

Furthermore, the following code (unrelated to the example below) seems to be particularly slow. Is there any way to speed this up? The inputs are numpy arrays.

```
def to_rgbd(frame, depth_frame, intrinsics, poses, device=None):
    frame = torch.tensor([[frame]], dtype=torch.float32)
    depth_frame = torch.tensor([[depth_frame]], dtype=torch.float32).unsqueeze(-1)
    intrinsics = torch.tensor([[intrinsics]], dtype=torch.float32)
    return RGBDImages(frame, depth_frame, intrinsics, poses, device=device)
```

## To Reproduce

```
from time import time
import torch
from torch.utils.data import DataLoader

from gradslam.datasets.icl import ICL
from gradslam.slam.pointfusion import PointFusion
from gradslam.structures.rgbdimages import RGBDImages
from gradslam.structures.pointclouds import Pointclouds

dataset_path = 'ICL'

if __name__ == "__main__":
    device = torch.device("cuda") if torch.cuda.is_available() else torch.device("cpu")
    dataset = ICL(dataset_path, seqlen=1, height=300, width=300)
    loader = DataLoader(dataset=dataset, batch_size=1)

    slam = PointFusion(device=device)
    pointcloud = Pointclouds(device=device)

    num_frames = 100
    time_elapsed = 0.0

    prev_frame = None

    for _ in range(num_frames):
        start = time()
        poses = torch.eye(4).view(1, 1, 4, 4) if prev_frame is None else None
        colors, depths, intrinsics, *_ = next(iter(loader))
        current_frame = RGBDImages(colors, depths, intrinsics, poses, device=device)
        pointcloud, current_frame.poses = slam.step(pointcloud, current_frame, prev_frame)
        prev_frame = current_frame
        time_elapsed += time() - start
    
    print('FPS', num_frames/time_elapsed)
```

## Environment

PyTorch version: 1.7.0
Is debug build: True
CUDA used to build PyTorch: 10.2
ROCM used to build PyTorch: N/A

OS: Ubuntu 18.04.5 LTS (x86_64)
GCC version: (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Clang version: Could not collect
CMake version: Could not collect

Python version: 3.6 (64-bit runtime)
Is CUDA available: True
CUDA runtime version: Could not collect
GPU models and configuration: GPU 0: GeForce RTX 2070 SUPER
Nvidia driver version: 450.80.02
cuDNN version: Could not collect
HIP runtime version: N/A
MIOpen runtime version: N/A

Versions of relevant libraries:
[pip3] numpy==1.19.2
[pip3] torch==1.7.0
[conda] Could not collect

krrish94, 2020-11-22:
Hi @apoorvkh,

Indeed, processing speeds and code optimization are some of our focus areas in subsequent releases. (and yes, decreasing `numiters` is expected to cause degradation in reconstruction quality)

One design choice we've made early on was to have gradslam (at least initial versions) research-ready. This included preferring pure pytorch code over (CUDA and/or JIT) among other calls made.

We're happy to have suggestions from the community on these aspects.

Regarding the specific codeblock you pointed to, the operation there is a CPU -> GPU memcpy and it is indeed a bottleneck. That said, there might indeed be ways to speed things up. We will review dataloading in the near-term. If you find anything, do not hesitate to get back to us

apoorvkh, 2020-11-24:
Thanks @krrish94! That would be good and I will let you know if I find any solutions in the near term.

closed 2023-04-30 by apoorvkh