gradslam/gradslam #24: KinectFusion support

SilvioGiancola, 2020-12-15:
Hi,

I would like to reproduce the performances of KinectFusion from your paper. Do you have reproducible code for it? If not, are you planning to release it?

Best,

krrish94, 2020-12-15:
@SilvioGiancola KinectFusion is indeed on our roadmap for version 0.2.0

SilvioGiancola, 2020-12-15:
Great, thank you for your prompt answer. Is there a candidate date for the release of version 0.2.0?

krrish94, 2020-12-15:
We're actively working on it -- although we want to get a sense of a stable API and feature requests first

My estimate would be first or second week of Jan

We will make our roadmap public later this week, and hopefully a lot of people will be eager to contrib :)

SilvioGiancola, 2020-12-15:
Thank you for your great work, looking forward to build upon it and contrib :)

cxt98, 2021-01-28:
Hi, is there any progress update on releasing KinectFusion?

krrish94, 2022-12-20:
Closing this, as we have been working on `PointFusion` as the sole candidate in learning applications (in part due to the prohibitive memory requirements imposed by `KinectFusion` on large scenes -- in the absense of sparse voxel-based hashing methods)

closed 2022-12-20 by krrish94