gradslam/gradslam #29: Its possible to use PointClouds?

lucasamparo, 2021-01-07:
## ❓ Questions and Help

Its possible to use just point clouds as input?
Without pose matrix or something like that?

I'm trying to register point clouds captured by an industrial laser scanner... but we don't have pose estimation for any point cloud, just the capture order.

krrish94, 2021-01-07:
@lucasamparo If I understand correctly, you're using mobile laser scanning (MLS) to capture pointclouds?

We currently have no support for lidar pointclouds of this kind -- this is a good issue for someone to work on / propose a feature.

lucasamparo, 2021-01-07:
Not a LiDAR, but something like this. Structured Light, the Newton Labs M500UW.

Do you have an idea how can I convert the point cloud to the input data for the framework?

krrish94, 2021-01-07:
That unfortunately won't be possible in the current framework. This will require implementing a pointcloud SLAM variant (which is extremely interesting and welcome contribs/discussion on, but don't have the cycles to do it ourselves). While we have the differentiable tools necessary to perform pointcloud alignment (gradICP), we would need to implement an `OdometryProvider` that will not need normals. We would also need to write a SLAM module that will fuse pointclouds to build a map (similar to the `ICPSLAM` and `PointFusion` modules.

In short, we'd need to modify the framework to take in pointclouds, not the other way around.

lucasamparo, 2021-01-07:
Hm. Ok. Thanks, anyway.

closed 2021-01-07 by lucasamparo