gradslam/gradslam #35: How to run gradslam with the TUM dataset?

zhanghua7099, 2021-03-04:
How to run gradslam with the TUM dataset? 

I want to run gradslam in TUM dataset. I try to run `examples/icpslam.py` with the command:

`python icpslam.py --dataset tum --dataset_path 'fr1's dataset'
`

But I just got two point clouds for visualization. 

Can I run gradslam like orbslam2 or other slam method?

NikV-JS, 2021-03-05:
Thanks for your interest in GradSLAM!
Yes, you can definitely run gradslam for whole scene reconstructions.
This previous query is very related to yours: #12 
I would actually point you to this demo code of  `gradslam PointFusion` on `TUM/rgbd_dataset_freiburg1_xyz`:
https://colab.research.google.com/drive/1uxjEJI5B2yJJDG3SxoAagH9uEuT67rXC?usp=sharing

If you are interested in whole scene reconstructions, you can have a look at our [gradslam documentation](https://gradslam.readthedocs.io/en/latest/) and the `slam.step()` function and tutorials would be of high interest. Ideally, you would want to set up the `gradslam tum dataloader` and run gradslam on batches to obtain whole scene reconstructions.

krrish94, 2021-04-14:
Closing, due to inactivity

closed 2021-04-14 by krrish94