gradslam/gradslam #38: Question about step by step method!

Yarp96, 2021-04-09:
## ❓ Questions and Help
Hi,

First, amazing paper and repo, its been super interesting to test this SLAM technique.

My question is about the basic Pointfusion and step by step Pointfusion. Is there any advantage of the basic technique or is it always better to use the step method? 

Best regards,

krrish94, 2021-04-14:
Thanks @Yarp96 !

The step-by-step Pointfusion approach is useful when you're using gradslam in an interactive application (e.g., a robot navigating around a room captures an image, decides on a next action, and receives another image upon successful execution of that action. The sequence of images depends on the actions chosen and are thus not available a priori)

However, the basic (batch) PointFusion approach can be used if all images in the sequence are available beforehand.

krrish94, 2022-12-20:
Closing due to inactivity

closed 2022-12-20 by krrish94