gradslam/gradslam #40: How are loop closures supported in GradSLAM?

paucarre, 2021-04-13:
## ❓ Questions and Help

Loop closures are not explicitly discussed in the original paper for GradSLAM nor are present in the project's code nor documentation in an explicit way. 
As loop closures play a fundamental role in classical SLAM, I'd like to know if you could explain how GradSLAM deals with them.

As far as I could understand from your paper, you use the "active part of the map" (  'Ma' )  from features, which they are matched with sensor data (Pa) in a one-to-many fashion. It is not clear to me if the position and orientation of the sensor plays a role in such matching (e.g. if odometry from the integration of velocities plays a role in sensor-to-map matching). 

Furthermore, the  "Differentiable map fusion" is to me the only thing that resembles loop closure as I guess one can consider that when the same object is seen twice, then they are fused ( "or loop closed" ). Is this the case?
Again, it is not really clear whether the position+orientation resulting from velocity integration plays any role there. 
The biggest concern I have is that if odometry is not taken into account, then two very similar observations could be easily "fused".

I would also like to know if "Differentiable map fusion" uses odometry and takes the role of loop closure, if it is reasonable to transform its inputs before entering in the "Differentiable map fusion" module so that a custom trained "loop closure detector" can be learnt from data.

krrish94, 2021-04-14:
Thanks for your interest @paucarre!

In gradslam, at each time step we align the "entire map" with the current frame. This implicitly means that we're performing global optimization at each step (and wouldn't need explicit loop closure detection). However, this also means that currently gradslam can operate in relatively-small environments (as global alignment at each timestep would become computationally expensive for large scenes).

krrish94, 2021-04-14:
The differentiable map fusion module indeed uses camera poses as input (which may in-turn be computed from odometry). This can be used for learning odometry estimation (but not loop closure detection, since we don't have an explicit loop closure step in gradslam)

paucarre, 2021-04-14:
Thanks for your answer @krrish94 !

That clarifies a lot.

I was wondering if you see feasible with the current state the following setup to solve the memory problem:

1. Using a neural network, downsample the high memory 3D textured map to a low-memory 2D map where 
it's fesible to do mapping with limited memory resources (which happens often in robotics) 

To be specific, say the original high-memory map we call "Detailed_Map" is a pointcloud (as per GradSLAM documentation).
Assume "Detailed_Map" pointcloud has the following tensor-size structure (as per documentation):
```
Detailed_Map = [
  points = [ (K_1, 3), (K_2, 3), ... , (K_N, 3) ],
  normals = [ (K_1, 3), (K_2, 3), ... , (K_N, 3) ],
  colors = [ (K_1, 3), (K_2, 3), ... , (K_N, 3) ]
]
```

where both `N` and `K_i` are very high (lots of samples with lots of points).

We want to reduce `N` and `K_i` in `Detailed_Map` by flattening and downsampling it to a 2D `Compact_Map` ("birds eye map").

A neural network would transform the original `Detailed_Map` to a `Compact_Map` with the following structure:
```
Compact_Map = [
  points = [ (H_1, 2), (H_2, 2), ... , (H_M, 2) ],
  normals = [ (H_1, 2), (H_2, 2), ... , (H_M, 2) ],
  embeddings = [ (H_1, L), (H_2, L), ... , (H_M, L) ]
]
```

Where:
 - `M` is very significantly smaller than `N` ( `M` << `N`).
 - `H_i` is very significantly smaller than `K_i` (`H_i` << `K_i` ).

Then we perform both `Differentiable Map` as well as `Map Fussion` on the `Compact_Map` with low memory as the 
tensor itself will be orders of magnitude smaller.

2. `Compact_Map` to `Detailed_Map` Correspondence

Then the only thing that's left is to establish a correspondence between `Compact_Map` and `Detailed_Map`.
As they are locally similar from a 2D point prespective (and ground projected normals), one could try to 
reconstruct the `Detailed_Map` from `Compact_Map` using a convolutional layer. Something similar to 
an autoencoder.

3. Production setup

An autoencoder-style neural network wourld be trained to downsample `Detailed_Map` to `Compact_Map` on a high-memory computer outside the robot. 
A 2D map ("birth eye map") would be created using `Compact_Map` and `GradSLAM` layers.
The `Compact_Map`  would be used in the robot for real-time localization. 

Overall, I was wondering if setups like that would be feasible within the current state of the project of there 
would be major changes in the framework itself of these to be possible. I'm concerned the embedding space would need to be somewhat regularized to behave like RGB colors.

closed 2021-04-20 by paucarre