gradslam/gradslam #48: Usage of multiple RGBD sensors as input

RuslanAgishev, 2021-10-01:
## ❓ Usage of multiple RGBD sensors as input

Thank you for making this great project open to the public!

I have a question regarding the possibility to run the provided SLAM pipelines using observations from different sensors.
For example, could you guide me through (if it is possible) how to use RGBD inputs from different cameras (say, mounted on a robot with known extrinsics and intrinsics)?

Ruslan

Willforcv, 2021-12-10:
did you use this with RGB sensor input?

RuslanAgishev, 2021-12-13:
I used the RGB-D data

krrish94, 2022-12-20:
Currently, this is a single-camera RGB-D SLAM pipeline. While there aren't any foreseeable plans to extend this to multiple cameras, it is indeed possible for each input image to have a different intrinsics matrix -- so technically, this'd require a re-write of a dataloader that can load the appropriate intrinsics matrix per image (and also transform the loaded poses via known extrinsics of other cameras).

Closing this; but feel free to re-open if there is a pressing need for this feature.

closed 2022-12-20 by krrish94