gradslam/gradslam #54: No matching distribution found for open3d==0.10.0.0

yangyi02, 2022-08-30:
I tried both 
pip install gradslam
and
pip install 'git+https://github.com/gradslam/gradslam.git' -q

Both gives the same error:

ERROR: Could not find a version that satisfies the requirement open3d==0.10.0.0 (from gradslam) (from versions: none)
ERROR: No matching distribution found for open3d==0.10.0.0

It seems the latest open3d is 0.15.0 now, is it possible to upgrade the open3d dependency?

yangyi02, 2022-12-20:
Here is the full error message I got.

pip install gradslam

Collecting gradslam
  Downloading gradslam-0.1.0-py3-none-any.whl (87 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 87.7/87.7 kB 4.2 MB/s eta 0:00:00
Collecting torch>=1.6.0
  Downloading torch-1.13.1-cp310-cp310-manylinux1_x86_64.whl (887.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 887.5/887.5 MB 1.3 MB/s eta 0:00:00
Collecting plotly>=4.10
  Downloading plotly-5.11.0-py2.py3-none-any.whl (15.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 15.3/15.3 MB 65.8 MB/s eta 0:00:00
ERROR: Could not find a version that satisfies the requirement open3d==0.10.0.0 (from gradslam) (from versions: 0.16.0)
ERROR: No matching distribution found for open3d==0.10.0.0

krrish94, 2022-12-20:
Hi @yangyi02,

I noticed this issue with newer python versions. The workaround, for now (until we push a PR), will be to install this from source (and not via pip). As @EdwardjkFeng points out (and based on my recent tests), latest versions of open3d work well too. 

So,
1. Clone the gradslam repo
2. Edit `requirements.txt` to remove the `==0.10.0` suffix for `open3d`
3. Uninstall any previous `gradslam` installations by running `pip uninstall gradslam`
4. Build from source by running `python setup.py build develop`

We are working towards a newer version release that addresses this issue, with other feature additions and support. This workaround should do well in the meantime.

yangyi02, 2022-12-20:
Great, thank you for the fast reply.

closed 2023-10-24 by yangyi02

RitaRitaki, 2024-10-14:
I am facing the same issue in Google Colab tutorials...beacuse of the open3d issue i am not installing gradslam i just git clone the repo...then i get the error 
cannot import name 'Pointclouds' from 'gradslam' (unknown location)
so i guess i doesnt recognise some locations in the gir repo?
Any help?