gradslam/gradslam #55: not compiled with GPU support error in chamfer.py when using gradicp

prashkmr, 2022-12-04:
Hi, Thanks for this repository. This is indeed helpful. 
I was trying for SLAM error calcuation using the package and ran into this error while using the provide(pcd1, pcd2) function in gradicp.py. 

The error is as follows:

Traceback (most recent call last):
  File "/hdd/scratch/prashant/code/diff-lidar-slam-dslr*/gans-DSLR.py", line 261, in <module>
    slam_error=Slam_error(groundtruth_list, recon_list)
  File "/hdd/scratch/prashant/code/diff-lidar-slam-dslr*/SLAM_ERROR.py", line 109, in Slam_error
    gnd_graph=find_poseGraph(gradslam.Pointclouds(gnd_scans,gnd_scan_normals))   #list of pytorh Tensor
  File "/hdd/scratch/prashant/code/diff-lidar-slam-dslr*/SLAM_ERROR.py", line 92, in find_poseGraph
    transform=odomprov.provide(pcd1,pcd2)
  File "/hdd/scratch/prashant/code/diff-lidar-slam-dslr*/gradslam/gradslam/odometry/gradicp.py", line 106, in provide
    transform, _ = point_to_plane_gradICP(
  File "/hdd/scratch/prashant/code/diff-lidar-slam-dslr*/gradslam/gradslam/odometry/icputils.py", line 498, in point_to_plane_gradICP
    A, b, chamfer_indices = gauss_newton_solve(
  File "/hdd/scratch/prashant/code/diff-lidar-slam-dslr*/gradslam/gradslam/odometry/icputils.py", line 200, in gauss_newton_solve
    _KNN = knn_points(src_pc, tgt_pc)
  File "/home/prashant/miniconda3/envs/pyt/lib/python3.9/site-packages/chamferdist/chamfer.py", line 266, in knn_points
    p1_dists, p1_idx = _knn_points.apply(
  File "/home/prashant/miniconda3/envs/pyt/lib/python3.9/site-packages/chamferdist/chamfer.py", line 162, in forward
    idx, dists = _C.knn_points_idx(p1, p2, lengths1, lengths2, K, version)
RuntimeError: Not compiled with GPU support.

I am not able to figure out the issue here and why is the error occuring. 
I installed gradslam using pip.
I tried on multiple servers, but the same issue.
Request you to help me out.
Eagerly awaiting a response.

krrish94, 2022-12-04:
Hi @prashkmr,

This might be because of a gpu / driver mismatch between the pip wheel and the server. For `chamferdist` (and `gradslam`), I would recommend installing from source. The last I tried this was yesterday, and it worked on an RTX3090

prashkmr, 2022-12-08:
I did try installing from the source using the "Recommnended" instruction on the home page. 
The GPU details are 
NVIDIA GeForce RTX 3090
Cuda details:  release 10.1, V10.1.243
However I used conda environment with torch version: 1.11.0.
However The same error occurs when I try running a pytorch code with the above package.

Can you specify the exact details of the configuration you are using for running. 
Are you using a conda or pip.
which pytorch version.  and if using conda which exact version of pytorch and cuda from pytorch website.
and cuda details.

Would be of great help.