gradslam/gradslam #59: Unable to use distortion in ConceptFusion

christian-rauch, 2024-01-31:
## 🐛 Bug / Unexpected behaviors

When using `distortion` in the `dataconfigs`, the `GradSLAMDataset` fails to apply the distortion as it tries to use the OpenCV function `cv2.undistort` on a PyTorch tensor.

See:
https://github.com/gradslam/gradslam/blob/59ca872e3d265ad09f63c4793d011fad67064452/gradslam/datasets/basedataset.py#L217-L231

In line 222 the colour image is converted from numpy to a torch tensor (`color = torch.from_numpy(color)`). If `self.distortion` is given (line 229), `cv2.undistort` is then applied on the torch tensor instead of the numpy image (line 231).

## To Reproduce

Steps to reproduce the behavior:

1. create a `dataconfigs` that contains a list of distortion coefficients in `distortion`:
  ```yaml
  camera_params:
    cx: 318.9576721191406
    cy: 247.3009033203125
    distortion:
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    fx: 607.2198486328125
    fy: 607.144287109375
    image_height: 480
    image_width: 640
    png_depth_scale: 1000
  dataset_name: icl
  ```
2. run `python run_feature_fusion_and_save_map.py --dataset-path [...] --sequence [...] --dataconfig-path [...]/my/dataset/icl.yaml`

```
  File "[...]/gradslam/datasets/basedataset.py", line 233, in __getitem__
    color = cv2.undistort(color, K, self.distortion)
cv2.error: OpenCV(4.8.1) :-1: error: (-5:Bad argument) in function 'undistort'
```

## Expected behavior

The `GradSLAMDataset` should apply the undistortion with the coefficients.

krrish94, 2024-04-05:
This is indeed a bug -- thanks for catching it!

Happy to accept a PR fixing this

christian-rauch, 2025-04-07:
> This is indeed a bug -- thanks for catching it!
> 
> Happy to accept a PR fixing this

@krrish94 See #61 for a fix.